szafranski/RH-ota #82: Smart-hotspot discussion

RogerBFPV, 2020-03-23:
In PM' with Pawel,  he was sharing a smart-hotspot concept with me.

The concept was: automaticlly configuring auto-hotspot
so if you are within your wifi range, it will connect, otherwise it will create hotspot

Something to keep in mind is how will the user know which IP address to point their browser to?   It is possible to reference the pi via http://hostname:5000 (default is raspberrypi), but maybe that will need to be configurable as well.  

It will also need to be tested the hostname resolution works in both modes (other hotspot, or pi as the hotspot).

RogerBFPV, 2020-03-23:
Most hotspots require an access code in order to connect. Thus I think we could provide the user several options:
option 1: connect to specific hotspot 
option 2: Server as the hotspot. 
option 3: dynamic:  attempt option 1, and if that fails, revert to option 2..

Note: in all  options hard wired ethernet should be available with DHCP.

just-david, 2020-03-23:
I've been using the auto-hotspot thing for about a year now.  I'm rather pleased with how simple and reliable it has been. 

The way it works, is that the owner has to pre-configure the wifi hotspot(s) that the timer can / will connect too.

At boot, the script runs and looks for known wifi in range.  If one is found, then the timer connects to it.    
If no known wifi is found, then the timer creates it's own network (by default ROTORHAZARD), with password configured by user:  default (PASSWORD).  

For my home wifi,  I was able to setup a local DNS entry (in my wifi router) to point to the right local address.  

prior to that, i had just bookmarked the local address after looking it up on the timer using an attached screen. 

it is also possible to find the local address by doing a MAC address search on the local network. 
https://raspberrypi.stackexchange.com/questions/13936/find-raspberry-pi-address-on-local-network
or:
https://github.com/adafruit/Adafruit-Pi-Finder/

The reason I did this, was that when I'm at home, I wanted to be able to connect to internet and my pi at the same time.  if it was always on hotspot mode I couldn't do that.   

When I am using the timer for races,  It just stays in hotspot mode.

szafranski, 2020-04-04:
@RogerBFPV any new thoughts? Asking cause we will implement that feature soon.

szafranski, 2020-04-06:
https://github.com/szafranski/RH-ota/issues/90

szafranski, 2020-04-10:
Closing for now. Comment here if needed to reopen

PR #115 (2020-04-05): Adding wizard for configuring automatic hotspot or always hotspot.
Fixes #90
Fixes #82
closed 2020-04-10 by szafranski