szafranski/RH-ota #101: Weird configuration dump

szafranski, 2020-03-29:
[image]
pi 4 is doubled
server version and RH_version is "doubled"

idk what is going on

szafranski, 2020-03-29:
gpio_reset - just added is also doubled

DavidJJ, 2020-03-30:
This is caused by us using the "config" variable to hold both the json value, and the "flag" value. 
I think we should just get rid of the duplicated "flag" values and just use the value in the original field

eg:  if pi_4_cfg:   
     blah blah

instead of  
if pi_4_cfg:
   pi_4_FLAG = True
...

if pi_4_FLAG:
   blah blah

szafranski, 2020-03-30:
No problem for me.

szafranski, 2020-04-01:
Will you change this?

szafranski, 2020-04-01:
And I think that we can abandon update_without_pdf. 

It requires us to have like whole another branch to same 2MB

szafranski, 2020-04-03:
No clue why there is "user" with default "racer" since it was changed to "pfabi" in debug mode and pi on raspberry pi. Tried on both: pi and Linux.

[image]

szafranski, 2020-04-03:
Now it shows correctly. Any clues?

szafranski, 2020-04-04:
Has to be investigated - causing bugs

szafranski, 2020-04-04:
Reload config in a program loop?

DavidJJ, 2020-04-04:
I think we have this under control now.

closed 2020-04-04 by DavidJJ