szafranski/RH-ota #104: Adding RTC option

sl33pydog, 2020-03-31:
Noticed that RH uses normal raspberry pi clock.  An RTC would allow for not having to worry about heat times not being correct.  I used the adafruit guide to add a ds3231 connected through i2c.

Would like to have this added to the options when installing RH.

https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time

You can add support for the RTC by adding a device tree overlay. Run

sudo nano /boot/config.txt

to edit the pi configuration and add whichever matches your RTC chip:

dtoverlay=i2c-rtc,ds1307
or
dtoverlay=i2c-rtc,pcf8523
or
dtoverlay=i2c-rtc,ds3231

to the end of the file

sudo apt-get -y remove fake-hwclock
sudo update-rc.d -f fake-hwclock remove
sudo systemctl disable fake-hwclock
Run sudo nano /lib/udev/hwclock-set and comment out these three lines:
#if [ -e /run/systemd/system ] ; then
# exit 0
#fi
Also comment out the two lines:
/sbin/hwclock --rtc=$dev --systz --badyear
/sbin/hwclock --rtc=$dev --systz

First run date to verify the time is correct. Plug in Ethernet or WiFi to let the Pi sync the right time from the Internet. Once that's done, run sudo hwclock -w to write the time, and another sudo hwclock -r to read the time

Once the time is set, make sure the coin cell battery is inserted so that the time is saved. You only have to set the time once assuming the coin cell doesn't die.

And you mentioned I should write this.
"# or use ntp time - Pawel will know what I mean"

szafranski, 2020-03-31:
It this model of RTC supported officially by RH?

sl33pydog, 2020-03-31:
> […](#)

It's mentioned in the issues page.  I think all three would be supported since once you change the raspberry pi clock from it's own system to the rtc the pi just sees a time and date and thus RH would see that same time.  I've checked with my install and it works after I did those extra steps.  The user however would need to specify ds1307, pcf8523 or ds3231.  My board by default will contain a ds3231 which I've tested on and offline with days in between testing.

szafranski, 2020-03-31:
So that feature is not described on any RH official place/site yet?

sl33pydog, 2020-03-31:
I thought it was at one point guess not.  Looks like it's still an issue/enhancement.  https://github.com/RotorHazard/RotorHazard/pull/134#issuecomment-580971576

I've verified that it works on a raspberry pi 3B using a ds3231.  It won't show up on the systems page but starting the server it'll show up as an i2c module. I would think after I tried it on other Pi's and from scratch if you implement this add on that we could close their issue.

szafranski, 2020-03-31:
Ok, I will implement that after we release first beta of OTA in few days. You will be responsible for testing cause I don’t have that clock. Ok?

sl33pydog, 2020-03-31:
> Ok, I will implement that after we release first beta of OTA in few days. You will be responsible for testing cause I don’t have that clock. Ok?

Definitely.  I'll make sure to build from scratch each time and use different Pi's.  Add if I need to I'll make sure to fix the led issue on the pi 3B like how you helped me.

szafranski, 2020-03-31:
Cool!

szafranski, 2020-04-01:
So ALL those lines should be commented out? Cause 3 are with # already and 2 are not:

> Run sudo nano /lib/udev/hwclock-set and comment out these three lines:
#if [ -e /run/systemd/system ] ; then
#exit 0
#fi
Also comment out the two lines:
/sbin/hwclock --rtc=$dev --systz --badyear
/sbin/hwclock --rtc=$dev --systz
> 

??

sl33pydog, 2020-04-01:
Yes.  I pulled those out of the guide.  I thought it was confusing at first but yes they all need #.  But hold off on this.  I could have gotten it wrong that RH doesn't do this since my RTC module hadn't been written to yet

szafranski, 2020-04-01:
Ok, sure. Write when you will resolve this :)

szafranski, 2020-04-08:
So is it solved? Can be implemented?

szafranski, 2020-04-26:
@sl33pydog - how is it going?

szafranski, 2020-08-12:
So in case of @sl33pydog won't make any contributions in that regard it will stay how it is.

sl33pydog, 2020-08-12:
Hey @szafranski sorry I've been really busy with my PCB and getting it to work with OTA as far as level shifters are concerned and the weird flashing that happens with pro minis.  But my latest prototype is still showing that RTC isn't implemented when I scratch build RH with OTA-RH.  I'll be writing out what I do to get it to work.  It follows the adafruit guide.

szafranski, 2020-08-12:
Ok. No problem! Wish U progress with your stuff