szafranski/RH-ota #108: fresh install of Rotorhazard does not work

DavidJJ, 2020-04-02:
We have an issue where the rotorhazard managment setup portion is trying to write the actual rotor hazard config file BEFORE rotorhazard is installed.   

This is what the order of operations needs to be:
1. Install RH-ota
1. configure RH-ota settings: the ones in updater-config.json
1. Install RotorHazard server (just download and extract)
1. NOW we can configure Rotorhazard's config.json file. 
1. Finish setup of Rotorhazard server. 

output:
=================

           CONFIGURATION

        Admin name:         admin
        Admin password:     rotorhazard
        RotorHazard port:   5000
        LED amount:         0
        LED pin:            10
        LED inverted:       False
        LED channel:        0
        LED panel rotate:   0
        LED rows inverted:  False
        LED DMA:            0
        LED frequency:      0
        Debug mode:         False
        CORS allowed:       *
        Serial ports:       []

        Please check. Confirm? [yes/change/abort]

yes
Traceback (most recent call last):
  File "update.py", line 437, in <module>
    main()
  File "update.py", line 433, in main
    main_menu(config)
  File "update.py", line 406, in main_menu
    rpi_update(config)
  File "/home/pi/RH-ota/rpi_update.py", line 286, in rpi_update
    main_window(config, config_flag, config_soft)
  File "/home/pi/RH-ota/rpi_update.py", line 225, in main_window
    conf_rh()
  File "/home/pi/RH-ota/conf_wizard_rh.py", line 258, in conf_rh
    config_now = do_config()
  File "/home/pi/RH-ota/conf_wizard_rh.py", line 235, in do_config
    write_json(rh_config, f"{home_dir}/RotorHazard/src/server/config.json")
  File "/home/pi/RH-ota/modules.py", line 201, in write_json
    with open(file_name, 'w') as open_file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/RotorHazard/src/server/config.json'

szafranski, 2020-04-03:
So we have to prevent the user from configuring before installing or what? There is that "not configured" line...

szafranski, 2020-04-06:
https://github.com/szafranski/RH-ota/issues/99 - related

szafranski, 2020-04-07:
Use another thread to track that issue

closed 2020-04-07 by szafranski