szafranski/RH-ota #109: i2c flashing failure

DavidJJ, 2020-04-02:
#######################################################################
    ###                                                                 ###
    ###                          RotorHazard                            ###
    ###                                                                 ###
    ###                      OTA Updater and Manager                    ###
    ###                                                                 ###
    #######################################################################    

                            CHOOSE FLASHING TYPE:
    
                    1 - Flash each node automatically - rec.
    
                    2 - Flash nodes individually
    
                    3 - First time flashing
    
                    4 - Show I2C connected devices
    
                    5 - Flash using GPIO reset pins - obsolete
        
                    e - Exit to main menu
            
1

			Flashing node 1 (reset with I2C address: 10)

on command sent
off command sent
on command sent
avrdude -v -p atmega328p -c arduino -P /dev/ttyS0 -b 57600 -U flash:w:/home/pi/RH-ota/firmware/stable/node_0.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyS0
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
^[^[avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

			Node 1 - flashed

szafranski, 2020-04-03:
You have to first use option number 3 in the flashing menu. Also - if right now you have firmware version with just serial enabled all the time it could not be flashed. 
Solution would be to use option number 3 from menu and disable/keep reset pins pushed on all nodes that are not being flashed in given moment - just for that first time.

DavidJJ, 2020-04-04:
fixed.

closed 2020-04-04 by DavidJJ