szafranski/RH-ota #118: I2C and boot/config not set during fresh install

sl33pydog, 2020-04-08:
Running beta OTA and did this on "P" version OTA. 
Running raspberry pi 3b+ .  Set config to stable.  When I do a fresh install on buster or buster lite I have to enable i2c and spi in raspi-config.  Also I have to edit boot/config.txt with:

dtparam=i2c_baudrate=75000
core_freq=250

szafranski, 2020-04-08:
So U used main_stable or master branch?

sl33pydog, 2020-04-08:
Main_stable

szafranski, 2020-04-08:
oh. thanks than, we will push new ota in few days. I will ask U to check it than if U want. Thanks!

szafranski, 2020-04-09:
Has to be fixed in "master" as well. Bug.

szafranski, 2020-04-09:
#121 related

szafranski, 2020-04-09:
is working now - should be tested

szafranski, 2020-04-09:
@sl33pydog - should be working now with master release

szafranski, 2020-04-13:
looks like it works on desktop and does not on lite. 
We should re-think system configuring part

szafranski, 2020-04-15:
Big chance that new configuration script in master will resolve this. Will be implemented today/tomorrow

szafranski, 2020-04-15:
It is now implemented in master branch. Yo I will know for sure if that interface was implemented or not for sure.

szafranski, 2020-04-26:
An also in stable now

szafranski, 2020-07-22:
Assume resolved

commit da3579fd (2020-04-09): minor tweaks
fixes #118
closed 2020-07-22 by szafranski