szafranski/RH-ota #161: attempting to list all UART connected devices crash.

DavidJJ, 2020-10-18:
FLASHING MENU
            
            
                    1 - Flash each node automatically      
    
                    2 - Flash nodes individually
    
                    3 - First time flashing
    
                    4 - Show I2C connected devices
                    
                    5 - Check all UART connected devices
            
                    e - Exit to main menu

            
5

		Checking node 1 (reset with I2C address: 10)

on command sent
off command sent
on command sent
avrdude -v -p atmega328p -c arduino -P /dev/ttyS0 -b 57600
Traceback (most recent call last):
  File "update.py", line 494, in <module>
    main()
  File "update.py", line 490, in main
    main_menu(config)
  File "update.py", line 469, in main_menu
    old_flash_gpio(config) if config.old_hw_mod else flashing_menu(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 568, in flashing_menu
    check_uart_devices(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 522, in check_uart_devices
    check_uart_connection() if not config.debug_mode else None
  File "/home/pi/RH-ota/nodes_flash.py", line 488, in check_uart_connection
    os.system(f"avrdude -v -p atmega328p -c arduino -P /dev/{config.port_name} -b 57600 || {show_uart_con_error_msg()}")
NameError: name 'config' is not defined

szafranski, 2020-10-18:
Thanks for reporting - I am thinking about fixing or removing that feature. Fix should be fairly easy - some importing from config module error/issue.

szafranski, 2020-10-19:
Fixed in PR #162

closed 2020-10-19 by szafranski