szafranski/RH-ota #165: Only One Node Confuses Node Reset

CerberusVelvet, 2020-11-02:
When using a build with just one node. 

Wiring the reset pin of Node1 to Pi GPIO pin 12 (GPIO18) as in documentation. Does not work for auto update or manual update. First update works. 

Wiring reset pin of Node1 to Pi GPIO pin 11 (GPIO17) enables manual update to work. Auto update still errors.

`Traceback (most recent call last):
  File "update.py", line 494, in <module>
    main()
  File "update.py", line 490, in main
    main_menu(config)
  File "update.py", line 469, in main_menu
    old_flash_gpio(config) if config.old_hw_mod else flashing_menu(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 560, in flashing_menu
    flash_firmware_onto_all_nodes(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 94, in flash_firmware_onto_all_nodes
    prepare_mate_node(addr) if not config.debug_mode else print("simulation mode - flashing disabled")
  File "/home/pi/RH-ota/nodes_flash_common.py", line 68, in prepare_mate_node
    bus.write_i2c_block_data(addr, reset_mate_node_command, on)
OSError: [Errno 121] Remote I/O error
`

szafranski, 2020-11-09:
Please check according to Facebook comment.

CerberusVelvet, 2020-11-09:
Nope, Same error.

szafranski, 2020-11-09:
Don’t want to sound patronizing but please make sure that U updated to
newest beta version of OTA, reopen it and than try. It works for me.

CerberusVelvet, 2020-11-09:
I misread the directions..... I updated the install (fresh download) but didn't update the install from inside the program.

Having done that, it works!

szafranski, 2020-11-09:
Cool, closing the issue. You can switch ack to main non-beta

closed 2020-11-09 by szafranski