szafranski/RH-ota #171: Node flash not working

DavidJJ, 2020-12-06:
Raspberry Pi 3b+
New beta board. 

```Flashing procedure started

		Flashing node 1 (reset with I2C address: 10)

Traceback (most recent call last):
  File "update.py", line 494, in <module>
    main()
  File "update.py", line 490, in main
    main_menu(config)
  File "update.py", line 469, in main_menu
    old_flash_gpio(config) if config.old_hw_mod else flashing_menu(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 563, in flashing_menu
    flash_firmware_onto_all_nodes(config)
  File "/home/pi/RH-ota/nodes_flash.py", line 97, in flash_firmware_onto_all_nodes
    prepare_mate_node(addr) if not config.debug_mode else print("simulation mode - flashing disabled")
  File "/home/pi/RH-ota/nodes_flash_common.py", line 68, in prepare_mate_node
    bus.write_i2c_block_data(addr, reset_mate_node_command, on)
OSError: [Errno 121] Remote I/O error
```

```
pi@rotorhazard:~/RH-ota $ cat version.txt 
2.5.0h1

Version of this program - it has nothing to do with the RH version.
It refers to the API level of newest contained nodes firmware.
For example version 2.2.10 contains firmware with API level 22.
Third number refers to actual version of the OTA updater itself.

Features and functions are described in files in /docs folder.

# keep version number in first line of this file

```
```

pi@rotorhazard:~/RH-ota $ cat updater-config.json 
{
    "pi_user": "pi",
    "rh_version": "master",
    "country": "US",
    "nodes_number": 4,
    "gpio_reset_pin": false,
    "port_name": "ttyS0",
    "debug_mode": false,
    "debug_user": "racer",
    "old_hw_mod": false,
    "pins_assignment": "default",
    "beta_tester": false
}pi@rotorhazard:~/RH-ota $ 
```
asdf

szafranski, 2020-12-06:
This error indicates no communication with the i2c device - node 2 in this
case. Check if i2c is enabled (try in sudo raspi-config, on-off-on and
reboot) and if node contains proper firmware.

szafranski, 2021-01-14:
Have U resolved this? It was surely due to no communication with I2C end-device.

szafranski, 2021-01-22:
@DavidJJ ?? ;)

DavidJJ, 2021-01-27:
Flashing now works.  part of the problem was having wrong value resistor installed on the board.  resister replaced with correct value and software now works.

closed 2021-01-27 by DavidJJ