szafranski/RH-ota #182: Feature request: Add port fowarding option

rodcar125, 2021-01-22:
Hi Pawel!

I have been port forwarding since I was aware of the function. Basically, instead of typing the RH ip address:port you just have to write the ip address.

The thing is that when adding port forwarding and you have raspAP installed/set up every time the pi is rebooted it deletes the forwarding rule. Also, the user might want to go to the raspAP dashboard which is shown when typing the ip address on the browser.

The request is to add two lines on the rc.local file so it runs every time the pi boots up and if the user wants to enter raspAP then comment those lines and reboot.

I suggest that the append of the lines is done at the end of the install process from OTA.

add following lines to rc.local before exit 0 line
sudo iptables -A PREROUTING -t nat -p tcp --dport 80 -j REDIRECT --to-ports 5000
sudo iptables-save

rc.local file would look something like
[image]

szafranski, 2021-01-22:
Hello! What would be the difference between that and changing port number in RotorHazard config to 80?

rodcar125, 2021-01-22:
if you set port 80 in the RH config file, then the system crashes... I guess this is because the web services are already being used as mentioned in Port Forwarding section  at https://github.com/RotorHazard/RotorHazard/blob/master/doc/Software%20Setup.md

If I use this config...
[image]

Then I get this error and RH server crashes...
[image]

szafranski, 2021-01-22:
Ok, thanks man! Will think about it. Could U ask about it on the RH group?
I would like to know peoples opinion about it.
Btw. - nice to hearing from U :)

rodcar125, 2021-01-22:
Yeah, been out for a while but I im back LOL.
Good to hear from you too!

even if I run it with sudo...
[image]

rodcar125, 2021-01-22:
I will raise an issue on the RH project as well. thanks man and all the best!

szafranski, 2021-03-03:
I will implement it when I will have an opportunity to use Raspberry Pi. For now I only use Banana Pi and I have to be sure that all features/commands work as expected. Idea is good - for sure.

szafranski, 2021-04-27:
In testing now. Will be added soon.

szafranski, 2021-05-03:
Done, added in 233.25.3i3

Will be in stable release very soon. 
Being added automatically during first installation of RH using OTA
or even during an update.

closed 2021-05-03 by szafranski

rodcar125, 2021-05-03:
Hi Pawel! Thanks so much, awesome addition in my opinion! 

Was this implemented with a option to enable/disable it how is it going to work?

thanks again! 
mall the best,
Rodrigo

szafranski, 2021-05-03:
So basically you are asking about an option to disable it?

rodcar125, 2021-05-03:
No no, of course not. Don’t get me wrong, I was the one raising the feature request. (: What I meant is that not everybody might want to redirect the port for whatever reason so i thought maybe the software should ask if the user wants to forward the port or not. Thank you again man!

szafranski, 2021-05-03:
Ok, ok, understood . I think I will leave it automatically like it is, but will add a feature to disable it. Ok?