szafranski/RH-ota #215: RH uses main not master for its dev version

martinglass, 2021-04-16:
In the lead-in page, OTA says its options are for "stable", "beta" and "master". 
_**Possible RotorHazard server versions that may be selected:

    > 'stable'- last stable release (can be from before few days or few months)

    > 'beta'  - last 'beta' release (usually has about few weeks, quite stable)

    > 'master'- absolutely newest features implemented (even if not well tested)**_

However, the selection pages says the options are "stable" "beta" and **"main"**

If the user types "master" ota tries to clone https://codeload.github.com/RotorHazard/RotorHazard/zip/master

This fails because RH doesn't have a branch called "master", even though RH-OTA has its main branch named as master (not main). Confusion needs to be cleared up.

Suggestion is to change the lead-in page to say "main", which is the same as written on the selection page. You may want to go further and rename the RH-OTA main branch to "main" too, instead of "master", to align with RH.

szafranski, 2021-04-17:
Thanks! So:
that: " 'master'- absolutely newest features implemented (even if not well tested)"
I changed to "main" - thanks for pointing that out.

> If the user types "master" ota tries to clone https://codeload.github.com/RotorHazard/RotorHazard/zip/master

I don't know how U achieved that cause normally, when U input "master" you will see:
[image]

Suggestion about changing RH-ota branches to be aligned with RotorHazard ones - already working on it, but it is not very crucial.

szafranski, 2021-04-17:
PR about RH-ota branch name change from 'master' to 'main' is pending now. 
Please explain that last thing about trying to clone wrong RH branch. Thanks.

szafranski, 2021-04-27:
I will close the issue soon in case of lack of info. Thanks

martinglass, 2021-04-27:
Maybe I was muddled. It was late at night when I was working on this. Please close this if you want.

szafranski, 2021-04-27:
Hah, no problem man. Thanks for reporting anyways!

closed 2021-04-27 by szafranski