szafranski/RH-ota #228: Updater keeps reporting it needs an update

propson, 2021-05-22:
I'm running Updater version 300.35.3i5.
Each time I run OTA I tells me it needs an update.
I then perform the update and it reports that it's updated correctly.
Then after a reboot it will again report it needs an update.

szafranski, 2021-05-22:
Thanks for reporting but i cannot replicate that. You mean the RH update or OTA itself update?

propson, 2021-05-22:
So this is what I see:
[image]

[image]

I then proceed to update using 'u' twice

The result will be:

`
Software will be automatically closed.

Word 'Killed' may be shown.

Please wait...

Enter 'sudo' password if prompted.
sudo: unable to resolve host timer: Temporary failure in name resolution

.Updating process has been started

.................................................

./ota.sh: line 144:  1324 Killed                  python3 start_ota.py
pi@timer:~/RH-ota $

        Please wait - checking internet connection state....

        Looks like you don't have internet connection.
 -- attempt to fix the nameserver --

# Generated by resolvconf
nameserver 127.0.0.1
changing to:
sudo: unable to resolve host timer: Temporary failure in name resolution
sudo: unable to resolve host timer: Temporary failure in name resolution
nameserver 1.1.1.1
old nameserver is saved as: /etc/resolv.conf.orig
sudo: unable to resolve host timer: Name or service not known

 -- nameserver fixed --

Trying again...

        Please wait - checking internet connection state....

                Internet connection - OK

         Please wait: updating process from version 300.35.3i5

..............................

Redirecting output to ‘wget-log’.
Archive:  tempota.zip
fa2e90a547966dbc44c3b7a3bcad86dd3965b49e
   creating: RH-ota-stable/
   creating: RH-ota-stable/.dev/
   creating: RH-ota-stable/.dev/backup_tools/
  inflating: RH-ota-stable/.dev/backup_tools/flash.sh
  inflating: RH-ota-stable/.dev/backup_tools/gpio_fix
  inflating: RH-ota-stable/.dev/downlog.py
  inflating: RH-ota-stable/.dev/i2c_emulator
  inflating: RH-ota-stable/.dev/i2c_test.py
  inflating: RH-ota-stable/.dev/isPi4.sh
  inflating: RH-ota-stable/.dev/pifind.py
   creating: RH-ota-stable/.dev/uart_diagnostics/
  inflating: RH-ota-stable/.dev/uart_diagnostics/Arduino_serial_code.ino
  inflating: RH-ota-stable/.dev/uart_diagnostics/uart_test_write_to_arduino.py
 extracting: RH-ota-stable/.first_time_here
  inflating: RH-ota-stable/.gitignore
[... 95 lines cut ...]
  inflating: RH-ota-stable/scripts/iptables_conf.sh
  inflating: RH-ota-stable/scripts/log_send.sh
  inflating: RH-ota-stable/scripts/log_write.sh
  inflating: RH-ota-stable/scripts/net_fixer.sh
  inflating: RH-ota-stable/scripts/rtc_sync.sh
  inflating: RH-ota-stable/scripts/self_updater.sh
  inflating: RH-ota-stable/scripts/server_start.sh
  inflating: RH-ota-stable/scripts/sys_conf.sh
  inflating: RH-ota-stable/scripts/update_rh.sh
  inflating: RH-ota-stable/scripts/updater_from_ota.sh
  inflating: RH-ota-stable/self_update.py
  inflating: RH-ota-stable/start_ota.py
  inflating: RH-ota-stable/update.py
  inflating: RH-ota-stable/version.txt

    RotorHazard OTA Manager updated to version 300.35.3i5

            You may check update-notes.

  Update process done, please re-enter ~/RH-ota directory
  by typing: cd ~/RH-ota

         -- Hit Enter to continue --

`

szafranski, 2021-05-22:
Please type:

sudo nano /etc/resolv.conf

append at the end of the file:

nameserver: 8.8.8.8

if there is any other "nameserver" already - remove that line or comment it with #

Than ctrl+S to save

reboot the pi

than tell me what happened

propson, 2021-05-22:
It was showing this:
# Generated by resolvconf
nameserver 127.0.0.1

After changing it to 8.8.8.8 it worked one time.
Then I rebooted and the file was back to it's original state.
Something is overwriting it.

I'll try a fresh install on a new SD card.

Thanks for the help.

szafranski, 2021-05-22:
Sure bud. I have no idea why U got those. Hit me if any further help needed.

szafranski, 2021-07-10:
Closing for now, cannot replicate. Write if problem occurs again. Thanks

closed 2021-07-10 by szafranski